universelabs/universe-key-manager #13: iPhone support

faddat, 2019-08-10:
**Describe the bug**
Doesn't work on iPhone 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

guylepage3, 2019-08-19:
Hey @faddat can you provide a description for this? Thanks.

guylepage3, 2019-09-16:
@faddat what is this issue? I don't understand it as it's just the issue template. Can we close this out?

faddat, 2019-09-23:
Feather was working on android, not iphone.  Closing this issue now.

closed 2019-09-23 by faddat